tinygo-org/tinygo #4866: "out of bounds memory access" in runtime.alloc / hashmapStringGet

jakebailey, 2025-04-22:
When running the TypeScript Go port (modified to remove JSON calls that aren't currently working in tinygo), I'm seeing this error when running the binary:

`wazero`:

```
error instantiating wasm binary: module[main] function[_start] failed: wasm error: out of bounds memory access
wasm stack trace:
	main.runtime.alloc(i32,i32) i32
		0x816c: /usr/local/lib/tinygo/src/runtime/gc_blocks.go:409:5
	main.runtime.hashmapStringGet(i32,i32,i32,i32,i32) i32
		0xe447: /usr/local/lib/tinygo/src/runtime/hashmap.go:518:35 (inlined)
		        /usr/local/lib/tinygo/src/runtime/arch_tinygowasm_malloc.go:1074:90
	main.github.com/microsoft/typescript-go/internal/scanner.GetIdentifierToken(i32,i32) i32
		0x1b9928: /home/jabaile/work/TypeScript-go/internal/scanner/scanner.go:1936:6 (inlined)
		          /home/jabaile/work/TypeScript-go/internal/core/compileroptions_stringer_generated.go:892:12 (inlined)
		          /home/jabaile/work/TypeScript-go/internal/scanner/scanner.go:26:19 (inlined)
```

`wasmtime`:

```
error instantiating wasm binary: module[main] function[_start] failed: wasm error: out of bounds memory access
wasm stack trace:
	main.runtime.alloc(i32,i32) i32
		0x816c: /usr/local/lib/tinygo/src/runtime/gc_blocks.go:409:5
	main.runtime.hashmapStringGet(i32,i32,i32,i32,i32) i32
		0xe447: /usr/local/lib/tinygo/src/runtime/hashmap.go:518:35 (inlined)
		        /usr/local/lib/tinygo/src/runtime/arch_tinygowasm_malloc.go:1074:90
	main.github.com/microsoft/typescript-go/internal/scanner.GetIdentifierToken(i32,i32) i32
		0x1b9928: /home/jabaile/work/TypeScript-go/internal/scanner/scanner.go:1936:6 (inlined)
		          /home/jabaile/work/TypeScript-go/internal/core/compileroptions_stringer_generated.go:892:12 (inlined)
		          /home/jabaile/work/TypeScript-go/internal/scanner/scanner.go:26:19 (inlined)
```

The repro for this takes a bit to get to (time wise, long clone, long build):

```console
$ git clone --branch jabaile/tinygo --recurse-submodules git@github.com:microsoft/typescript-go.git
$ GOOS=wasip1 GOARCH=wasm tinygo build -o tsgo-tinygo.wasm ./cmd/tsgo

$ wazero run -mount=/:/ -env-inherit ./tsgo-tinygo.wasm -p ~/work/TypeScript/src/compiler --singleThreaded
# or
$ WASMTIME_BACKTRACE_DETAILS=1 wasmtime run --dir=/ --env PWD="$PWD" --env PATH="$PATH" -W max-wasm-stack=1048576 ./tsgo-tinygo.wasm -p ~/work/TypeScript/src/compiler --singleThreaded
```

dgryski, 2025-04-23:
Can you try running with a larger stack size to see if it's stack-overflow-related memory corruption?

dgryski, 2025-04-23:
Also if this binary works on when compiled for native but not on wasm that would also be useful to know.

jakebailey, 2025-04-23:
I asked wasmtime for a max stack size of `1073741824`, and still got the error. IIRC wazero has a default stack limit on the order of 250MB, same as 32-bit Go programs do by default, so I think does itself show that the stack limit isn't the problem.

The binary works when run natively (plain `go build`), yes, as well as when using Go to build the wasm to run instead of tinygo.

aykevl, 2025-04-28:
> IIRC wazero has a default stack limit on the order of 250MB, same as 32-bit Go programs do by default, so I think does itself show that the stack limit isn't the problem.

TinyGo has its own internal stack management in linear memory, which for wasm is 64kB by default. That's usually enough, but not always (and can result in weird bugs). Hence why it's important to check.

> The binary works when run natively (plain `go build`), yes, as well as when using Go to build the wasm to run instead of tinygo.

What about when you use just `tinygo build`? No `-target=wasm` or anything, just build a native binary using TinyGo for your current OS. Things are often easier to debug when running TinyGo programs outside WebAssembly.

jakebailey, 2025-04-28:
> What about when you use just `tinygo build`? No `-target=wasm` or anything, just build a native binary using TinyGo for your current OS. Things are often easier to debug when running TinyGo programs outside WebAssembly.

Ah, it does in fact crash there. But, I don't think it's the same crash given the Wasm binary doesn't complain about GC/heap size?

```
GC Warning: Failed to expand heap by 137438969860 KiB
GC Warning: Failed to expand heap by 137438953476 KiB
GC Warning: Out of Memory! Heap size: 478 MiB. Returning NULL!
panic: runtime error at 0x000000000053e2e1: caught signal SIGSEGV
[1]    823859 segmentation fault (core dumped)  ./tsgo-tinygo -p $PWD/_submodules/TypeScript/src/compiler --singleThreaded
```

The Boehm PR crashes too, but differently too:

```
panic: runtime error at 0x00000000005477bf: caught signal SIGSEGV
[1]    826061 segmentation fault (core dumped)  ./tsgo-tinygo-boehm -p $PWD/_submodules/TypeScript/src/compiler 
```

dgryski, 2025-05-12:
This appears to just be a regular goroutine stack overflow.
```
~/go/src/github.com/microsoft/typescript-go $ tinygo build -target=wasip1 -stack-size=1024KB -tags=runtime_asserts -o tsgo-tinygo.wasm ./cmd/tsgo
~/go/src/github.com/microsoft/typescript-go $ WASMTIME_BACKTRACE_DETAILS=1 wasmtime run --dir=/ --env PWD="$PWD" --env PATH="$PATH" -W max-wasm-stack=1048576 ./tsgo-tinygo.wasm -p $PWD/_submodules/TypeScript/src/compiler --singleThreaded
panic: reflect: unimplemented: AssignableTo with interface
Error: failed to run main module `./tsgo-tinygo.wasm`

Caused by:
    0: failed to invoke command default
    1: error while executing at wasm backtrace:
           0:  0x34eb0 - runtime.abort
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/runtime_tinygowasm.go:78:6
                       - runtime.panicOrGoexit
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/panic.go:81:7
           1:   0x6d8d - runtime._panic
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/panic.go:55:15
           2:   0x8e7a - (*internal/reflectlite.RawType).AssignableTo
           3:  0xa386a - (*internal/reflectlite.RawType).Implements
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/internal/reflectlite/type.go:758:23
                       - (*reflect.rawType).Implements
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/reflect/type.go:428:29
                       - (Go interface method)
                           at <Go interface method>

```

When building with a 1M goroutine stack, we get the expected reflect issue.  ( https://github.com/tinygo-org/tinygo/issues/4277 ).

jakebailey, 2025-12-16:
Sure; I haven't had time to retry tinygo on tsgo.

closed 2025-12-16 by jakebailey