tinygo-org/tinygo #4867: "panic: runtime error: nil pointer dereference" in task.Pause

jakebailey, 2025-04-22:
`wazero`:

```
panic: runtime error: nil pointer dereference
error instantiating wasm binary: module[main] function[_start] failed: wasm error: unreachable
wasm stack trace:
	main.runtime.runtimePanicAt(i32,i32)
		0x1c1e4: /usr/local/lib/tinygo/src/runtime/runtime_tinygowasm.go:78:6 (inlined)
		         /usr/local/lib/tinygo/src/runtime/arch_tinygowasm_malloc.go:75:27 (inlined)
		         /usr/local/lib/tinygo/src/runtime/algorithm.go:38:37 (inlined)
		         /usr/local/lib/tinygo/src/runtime/arch_tinygowasm_malloc.go:1074:90
	main.runtime.nilPanic()
		0x732c: /usr/local/lib/tinygo/src/runtime/panic.go:180:16
	main.internal/task.Pause()
		0x170f4: /usr/local/lib/tinygo/src/internal/task/task_asyncify.go:96 (inlined)
		         /usr/local/lib/tinygo/src/internal/task/queue.go:1074:90
	main.runtime.deadlock()
		0x1d631: /usr/local/lib/tinygo/src/runtime/scheduler_cooperative.go:47:12 (inlined)
		         /usr/local/lib/tinygo/src/runtime/arch_tinygowasm_malloc.go:75:27 (inlined)
		         /usr/local/lib/tinygo/src/runtime/algorithm.go:38:37 (inlined)
		         /usr/local/lib/tinygo/src/runtime/arch_tinygowasm_malloc.go:1074:90
	main.(*github.com/microsoft/typescript-go/internal/core.parallelWorkGroup).Queue$1$gowrapper(i32)
		0x10e9e7: /home/jabaile/work/TypeScript-go/internal/core/workgroup.go:37:2 (inlined)
		          /home/jabaile/work/TypeScript-go/internal/core/compileroptions_stringer_generated.go:706:16 (inlined)
		          /home/jabaile/work/TypeScript-go/internal/core/compileroptions_stringer_generated.go:680:16 (inlined)
		          /home/jabaile/work/TypeScript-go/internal/core/compileroptions_stringer_generated.go:619:11 (inlined)
		          /home/jabaile/work/TypeScript-go/internal/core/compileroptions_stringer_generated.go:1129:34 (inlined)
		          /home/jabaile/work/TypeScript-go/internal/core/pattern.go:164:24 (inlined)
```

`wasmtime`:

```
panic: runtime error: nil pointer dereference
Error: failed to run main module `./tsgo-tinygo.wasm`

Caused by:
    0: failed to invoke command default
    1: error while executing at wasm backtrace:
           0: 0x1e562 - runtime.abort
                           at /usr/local/lib/tinygo/src/runtime/runtime_tinygowasm.go:78:6              - runtime.runtimePanicAt
                           at /usr/local/lib/tinygo/src/runtime/panic.go:103:7
           1: 0x96aa - runtime.nilPanic
                           at /usr/local/lib/tinygo/src/runtime/panic.go:180:16
           2: 0x19472 - internal/task.Pause
                           at /usr/local/lib/tinygo/src/internal/task/task_asyncify.go:96:0
           3: 0x1f9af - runtime.deadlock
                           at /usr/local/lib/tinygo/src/runtime/scheduler_cooperative.go:47:12
           4: 0x110d65 - <goroutine wrapper>
                           at /home/jabaile/work/TypeScript-go/internal/core/workgroup.go:37:2
           5: 0x241d - tinygo_launch
                           at /usr/local/lib/tinygo/src/internal/task/task_asyncify_wasm.S:59:0
           6: 0x1f4d3 - (*internal/task.Task).Resume
                           at /usr/local/lib/tinygo/src/internal/task/task_asyncify.go:114:17              - runtime.scheduler
                           at /usr/local/lib/tinygo/src/runtime/scheduler_cooperative.go:230:11              - runtime.run
                           at /usr/local/lib/tinygo/src/runtime/scheduler_cooperative.go:256:11              - _start
                           at /usr/local/lib/tinygo/src/runtime/runtime_wasmentry.go:20:5
    2: wasm trap: wasm `unreachable` instruction executed
```

This code is:

```go
func Pause() {
	if *currentTask.state.canaryPtr != stackCanary {
		runtimePanic("stack overflow")
	}

	currentTask.state.unwind()
}
```

Somewhere on that first line, there's a `nil`.

Same repro as #4866, but without the `--singleThreaded` flag passed.

```console
$ git clone --branch jabaile/tinygo --recurse-submodules git@github.com:microsoft/typescript-go.git
$ GOOS=wasip1 GOARCH=wasm tinygo build -o tsgo-tinygo.wasm ./cmd/tsgo

$ wazero run -mount=/:/ -env-inherit ./tsgo-tinygo.wasm -p $PWD/_submodules/TypeScript/src/compiler --singleThreaded
# or
$ WASMTIME_BACKTRACE_DETAILS=1 wasmtime run --dir=/ --env PWD="$PWD" --env PATH="$PATH" -W max-wasm-stack=1048576 ./tsgo-tinygo.wasm -p $PWD/_submodules/TypeScript/src/compiler --singleThreaded
```

dgryski, 2025-04-23:
@jakebailey Can you try running your example with a larger stack and see it the problem goes away?  We have insufficient detection for stack-overflow-caused memory corruption.

jakebailey, 2025-04-23:
I asked wasmtime for a 1GB stack limit, and it resulted in the same error.

dgryski, 2025-05-12:
This is the same goroutine stack limit issue as https://github.com/tinygo-org/tinygo/issues/4866 .  It's not wasmtime that needs the increased stack limit, but the tinygo compiler needs to create larger stacks for the goroutines.  This is accomplished with the tinygo build flag `-stack-size=1M`.

dgryski, 2025-12-16:
Can we close this?  I think the original question was answered.

jakebailey, 2025-12-16:
Sure 😄

closed 2025-12-16 by jakebailey