tinygo-org/tinygo #4872: Implement a new OS

nitwhiz, 2025-04-25:
Hi there,

I thought about porting tinygo to a custom OS (just because it's possible - at least I think so). It's running on MIPS and the tinygo compiler would be used to generate ELF files. I want to use C-bindings for everything, which I provide (header files, objects, ...).

I know this is a very vague question, but I don't find any information about which steps need to be done to implement a new OS and it's actually not much more than an idea for now.

b0ch3nski, 2025-04-28:
I've had this on my research list: https://github.com/icexin/eggos

nitwhiz, 2025-05-02:
So I had to realize this is way over my head (this issue is a classic Dunning-Kruger haha), and for now I personally won't persue this further.

Thanks for the resources, though! Definitely going to come back to this one!

closed 2025-05-02 by nitwhiz