tinygo-org/tinygo #4880: Runtime error "Server error: Netdev not set"

Mossaka, 2025-05-01:
I could compile [a MCP server](https://github.com/mark3labs/mcp-go/tree/main/examples/everything) using mcp-go to wasip2 target, with the [wasi-http-go](https://github.com/ydnar/wasi-http-go) interface and disabling the stdio transport due to #4865. However, when I run this component on wasmtime with wasi-http capability, I got a runtime error saying "server error: netdev not set". Would love some help on figuring out why is a netdev even being used in the first place. 

```console
wasmtime run -Shttp main.wasm
2025/05/01 19:25:03 SSE server listening on :8080
2025/05/01 19:25:03 Server error: Netdev not set
```

https://github.com/tinygo-org/net/blob/main/netdev.go#L147

dgryski, 2025-05-02:
What did you do to get the example building?

(I've fixed the `os/signal` issues but adding `wasi-http-go` I get
```
~/go/src/github.com/mark3labs/mcp-go/examples/everything $ tinygo build -target=wasip2
error: failed to encode a component from module

Caused by:
    0: failed to decode world from module
    1: module was not valid
    2: module requires an import interface named `wasi:http/types@0.2.0`
`wasm-tools component new` failed: exit status 1
~/go/src/github.com/mark3labs/mcp-go/examples/everything $ git diff .
diff --git a/examples/everything/main.go b/examples/everything/main.go
index fa1c304..90f69ec 100644
--- a/examples/everything/main.go
+++ b/examples/everything/main.go
@@ -9,6 +9,8 @@ import (

        "github.com/mark3labs/mcp-go/mcp"
        "github.com/mark3labs/mcp-go/server"
+
+       _ "github.com/ydnar/wasi-http-go/wasihttp"
 )

 type ToolName string
```

Mossaka, 2025-05-03:
Sorry, you need a WIT file for wasi-http. I've pushed my changes to a [forked branch](https://github.com/Mossaka/mcp-go-fork/tree/wasi-http/examples/everything) and you may see the diff [here](https://github.com/mark3labs/mcp-go/compare/main...Mossaka:mcp-go-fork:wasi-http). To run it, run the following command

```Console
cd examples/everything
tinygo build -target=wasip2.json main.go
wasmtime run -Shttp main.wasm
```

where wasip2.json is 
```json
{
        "inherits": ["wasip2"],
        "wit-package": "./wit/proxy.wit",
        "wit-world": "go:http/proxy",
        "emulator": "wasmtime serve -Scli --dir={tmpDir}::/tmp {}"
}
```

dgryski, 2025-05-08:
That is not working for me:
```
~/go/src/github.com/mark3labs/mcp-go/examples/everything $ git show |head -1
commit 78fa3977d55f5f3cc2e3a6e6225833af122c0915
~/go/src/github.com/mark3labs/mcp-go/examples/everything $ tinygo build -target=wasip2.json main.go
error: expected `world`, `interface` or `use`, found keyword `package`
     --> ./wit/proxy.wit:38:1
      |
   38 | package wasi:io@0.2.0 {
      | ^------
`wasm-tools component embed` failed: exit status 1
```

Mossaka, 2025-05-12:
Do you have the same exact WIT file as [this one](https://github.com/mark3labs/mcp-go/compare/main...Mossaka:mcp-go-fork:wasi-http#diff-f0130e63d0fa0cfc6b2cec337be5d13be9c8eb18f5cc77d265c751de7ea94071)? Could you please check your wasm-tools version?

dgryski, 2025-05-12:
Ok, I'm able to reproduce this now.

ydnar, 2025-05-12:
I guess we could add sockets to wasip2