tinygo-org/tinygo #4886: Can't reach low power mode on NRF52840 / nicenano / xiao ble, with TinyGo

FlorianSoler, 2025-05-05:
Hi,

I've been trying to get the nRF52840 into a proper low-power or deep sleep mode using TinyGo, but I'm running into problems. The [[official low power docs](https://tinygo.org/docs/concepts/low-power/)](https://tinygo.org/docs/concepts/low-power/) mention using:

```go
for {
    arm.Asm("wfi")
}
```

But when I measure current, it's still around **1.2 mA**, which is way too high for a chip like the nRF52 that should be in the µA range when idle.

I also tried directly writing to the low power register:

```go
func deepSleep() {
    const TASKS_LOWPWR_ADDR = 0x4000007C
    taskLowPower := (*volatile.Register32)(unsafe.Pointer(uintptr(TASKS_LOWPWR_ADDR)))
    taskLowPower.Set(1)
    device.Asm("wfi")
}
```

Still no improvement.

To verify, I ran similar logic on the same hardware using Arduino + Bluefruit and got proper low power results:

```cpp
#include <Arduino.h>
#include <bluefruit.h>

void setup() {
    sd_power_mode_set(NRF_POWER_MODE_LOWPWR);
    sd_power_dcdc_mode_set(NRF_POWER_DCDC_ENABLE);
    __WFE();
    __WFI();
    sd_app_evt_wait();
    NRF_POWER->SYSTEMOFF = 1;
}

void loop() {}
```

So the issue seems specific to TinyGo. Any working examples or advice on how to properly enter deep sleep on nRF52840 using TinyGo would be great.

Thanks!

eliasnaur, 2025-05-05:
Not an expert, but don't you also have to shut down any hardware blocks such as UART, USB etc.?

FlorianSoler, 2025-05-05:
> Not an expert, but don't you also have to shut down any hardware blocks such as UART, USB etc.?

Hi, i have tried : 

### 1. High Power Consumption (\~800µA):

Even after shutting down as many peripherals as possible, I am still getting around **800µA** of current consumption, which seems higher than expected for deep sleep. Below is the code I used to disable peripherals and attempt to go into a low power state:

```go
package main

import (
	"device/arm"
	"device/nrf"

	"machine"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	led.Low()

	// Turn off UART0
	nrf.UARTE0.ENABLE.Set(0)

	// Turn off ADC
	nrf.SAADC.ENABLE.Set(0)

	// Stop timers
	nrf.TIMER0.TASKS_STOP.Set(1)
	nrf.TIMER1.TASKS_STOP.Set(1)
	nrf.TIMER2.TASKS_STOP.Set(1)

	// Disable RNG
	nrf.RNG.TASKS_STOP.Set(1)

	nrf.SPIM0.ENABLE.Set(0)
	nrf.SPIM1.ENABLE.Set(0)
	nrf.SPIM2.ENABLE.Set(0)

	nrf.CLOCK.TASKS_HFCLKSTOP.Set(1)

	for {
		arm.Asm("wfi")
	}
}
```

Despite this, the current consumption stays around **800µA**, which is still quite high for an idle state.

### 2. Unable to Wake Up from `SYSTEMOFF` Mode with GPIO Interrupt:

I’ve also tried putting the device into **SYSTEMOFF** mode and waking it up using a GPIO interrupt, but I am unable to get it to wake up. Here’s the code I used to enter **SYSTEMOFF** mode:

```go
package main

import (
	"device/nrf"
	"machine"
	"time"
)

var (
	modeChan = make(chan bool, 1)
	blink    = false
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	led.High()
	time.Sleep(time.Second)
	led.Low()

	for {
		select {
		case newMode := <-modeChan:
			blink = newMode
		default:
		}

		if blink {
			led.High()
			time.Sleep(250 * time.Millisecond)
			led.Low()
			time.Sleep(250 * time.Millisecond)
		}

		if !blink {
			time.Sleep(5 * time.Second)
			enterSystemOff()
		}
	}
}

func enterSystemOff() {
	// Set the GPIO pin 10 to wake up on falling edge
	nrf.P0.PIN_CNF[10].Set(nrf.GPIO_PIN_CNF_SENSE_Low)

	// Put the device into SYSTEMOFF mode
	nrf.POWER.SYSTEMOFF.Set(1)
}
```

However, I am unable to wake the device back up after it enters **SYSTEMOFF**. The GPIO interrupt does not seem to trigger a wake-up, and the device stays in **SYSTEMOFF**.

Thank you for your help!

eliasnaur, 2025-05-07:
Again, no expert here, so just a few questions to consider

- Is your equipment designed to measure current in the µA range?
- Are you measuring the processor current, or the incoming (total) current to the board? The latter includes voltage regulators etc. Voltage regulators are not 100% efficient, particularly at low currents.

FlorianSoler, 2025-05-11:
> * Are you measuring the processor current, or the incoming (total) current to the board? The latter includes voltage regulators etc. Voltage regulators are not 100% efficient, particularly at low currents.

I solved an issue I had, which was caused by a misunderstanding of how the nRF's internal peripherals and sleep mode work with TinyGo.  

I managed to reduce the power consumption from **1.2mA / 890µA down to 10µA**, while still keeping interrupts enabled.  

The solution involved:
- Disabling unused peripherals like UART and SPI
- Enabling the low-frequency clock
- Disabling USB (USBD)

---

## Working Code

```go
package main

import (
	"device/arm"
	"device/nrf"
	"machine"
	"time"
)

const (
	buttonPin = machine.Pin(10)
	psuPin    = machine.Pin(13)
	led       = machine.LED
)

var (
	modeChan = make(chan bool, 1)
	blink    = false
)

func main() {
	// Configure LED pin
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	led.High()

	psuPin.Configure(machine.PinConfig{Mode: machine.PinOutput})
	psuPin.Low()

	buttonPin.Configure(machine.PinConfig{Mode: machine.PinInputPulldown})
	DisableAll()

	time.Sleep(time.Second)
	led.Low()

	// Set up interrupt on button pin
	buttonPin.SetInterrupt(machine.PinFalling, handleInterrupt)

	for {
		select {
		case newMode := <-modeChan:
			blink = newMode
		default:
			// No new mode received; continue with current mode
		}

		if blink {
			led.High()
			time.Sleep(250 * time.Millisecond)
			led.Low()
			time.Sleep(250 * time.Millisecond)
		} else {
			arm.Asm("wfe")
		}
	}
}

func DisableAll() {
	nrf.UARTE0.ENABLE.Set(0)

	// Turn off ADC
	nrf.SAADC.ENABLE.Set(0)

	// Stop timers
	nrf.TIMER0.TASKS_STOP.Set(1)
	nrf.TIMER1.TASKS_STOP.Set(1)
	nrf.TIMER2.TASKS_STOP.Set(1)

	// Disable RNG
	nrf.RNG.TASKS_STOP.Set(1)

	nrf.SPIM0.ENABLE.Set(0)
	nrf.SPIM1.ENABLE.Set(0)
	nrf.SPIM2.ENABLE.Set(0)

	nrf.CLOCK.TASKS_HFCLKSTOP.Set(1)
	nrf.USBD.ENABLE.Set(0) // - 850µA
}

func handleInterrupt(p machine.Pin) {
	select {
	case modeChan <- !blink:
	default:
	}
}

Thank you for asking the right questions and guiding me toward the right reflections.

closed 2025-05-11 by FlorianSoler