tinygo-org/tinygo #4887: Output to serial monitor is 'loosing' some characters (RP2350, maybe other platforms)

SnapshotCiTy, 2025-05-05:
Output to serial monitor is 'loosing' some characters

Text that is output fast by successive fmt.Println, fmt.Printf, print, println statements, result after a few lines in 'randomly missing' characters.

```Golang
func main() {
	time.Sleep(time.Second)

	startTime := time.Now()
	fmt.Printf("\033[2J")
	fmt.Printf("\033[H")

	numLines := 600
	delayMicroseconds := 0

	for range numLines {
		fmt.Println("12345678901234567890123456789012345678901234567890123456789012345678901234567890")

		time.Sleep(time.Microsecond * time.Duration(delayMicroseconds))
	}

	fmt.Printf("in %v - Printed %d lines with delay %d microseconds", time.Since(startTime), numLines, delayMicroseconds)
	time.Sleep(time.Second * 2)
	}
}
```

When run on a pico2-w and pimoroni tiny2350 (only boards I have to test this), results in some characters to be 'lost' and not output trough the serial interface. Tested with tinygo monitor, SerialTools (macOs) and Screen (also on macOs).

If we add a delay of a few microseconds before printing the next line, this behaviour stops and no characters are lost, if the delay is enough.

- When using fmt.Println - or fmt.Printf("...80 cols of text...\n") it requires a delay of 11 or 12 micro seconds to not 'loose characters'. 
- When printing with print or  println, the required delay is 24 to 25 micro seconds, any less and some characters are 'lost'.

Compiling with "-opt 2" or "-gc leaking" did not change this behaviour.

Running the program on macOS with "go run ." and "tinygo run ." doesn't exhibit this behavior.

Gustavomurta, 2025-06-30:
Trying to help you by testing Windows 11 PC.
Smallest "delayMicroseconds" value was 5, without overlapping bytes

==============================================================
/
Raspberry Pico 2 W - RP2350 - serial monitor test
Gustavo Murta 2025/06/29
tinygo version 0.38.0 windows/amd64 (using go version go1.24.2 and LLVM version 19.1.2)
tinygo flash -target pico2 main.go

https://github.com/tinygo-org/tinygo/issues/4887
Output to serial monitor is 'loosing' some characters

package main

import (
	"fmt"
	"time"
)

const (
	numLines          = 10
	delayMicroseconds = 2
)

func main() {

	time.Sleep(time.Second * 3) // Delay a bit on startup to easily catch the first messages
	println("Pico 2W Serial monitor test")

	fmt.Printf("\033[2J") //  What is this?
	fmt.Printf("\033[H")  //  What is this?
	println()
	startTime := time.Now()

	for range numLines {
		fmt.Println("1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890")
		time.Sleep(time.Microsecond * delayMicroseconds) // * time.Duration(delayMicroseconds)
	}

	fmt.Printf("Printed %d lines in %v with delay %d microseconds", numLines, time.Since(startTime), delayMicroseconds)
	time.Sleep(time.Second * 2)
}

Gustavomurta, 2025-06-30:
Pico 2W Serial monitor test
[2J[H
1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890
1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890
1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890
1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890
1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890
1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890
1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890
1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890
1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890
1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890
Printed 10 lines in 764µs with delay 5 microseconds

Gustavomurta, 2025-06-30:
Checking the time to print a line on Raspberry Pico RP240. 
Raspberry Pico 2W RP2350 does not support rp.TIMER yet.

package main

import (
	"device/rp"
	"fmt"
	"time"
)

const (
	numLines          = 10
	delayMicroseconds = 20
)

func main() {

	time.Sleep(time.Second * 3) // Delay a bit on startup to easily catch the first messages
	println("Pico 2W Serial monitor test")

	startTime := time.Now()

	for range numLines {
		start := rp.TIMER.TIMELR.Get()
		fmt.Println("1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890 1234567890")
		end := rp.TIMER.TIMELR.Get()
		fmt.Println("Duration time to print this line = ", (end - start), "us")
		time.Sleep(time.Microsecond * delayMicroseconds) // * time.Duration(delayMicroseconds)
	}

	fmt.Printf("Printed %d lines in %v with delay %d microseconds", numLines, time.Since(startTime), delayMicroseconds)
	time.Sleep(time.Second * 2)
}

squirrel-bob, 2025-07-14:
AFAIK that's the Pico/Pico 2 hardware issue - the small transmit buffer results in characters being lost.

Gustavomurta, 2025-07-14:
I don't think it's a hardware issue. I've never seen anything like that.
It sounds like a USB port buffer clearing issue to me.

Gustavomurta, 2025-07-14:
I have experience with the USB protocol for device-to-pc communication. On PCs (Windows and Linux), I've used the **Gousb package** (Google) extensively. In this case, I recommend using **Wireshark** to capture more details about the USB communication.

squirrel-bob, 2025-07-14:
What an unimaginable coincidence then, I guess. I personally find switching to a more mature MCU/board to be faster and easier than debugging RPI issues.

SnapshotCiTy, 2025-07-15:
@Gustavomurta what is your last example showing? that using the UART0 doesn't exhibit the issue? And that it only occurs using the hardware usb port? Can you try with 1000 lines? (I'm travelling and away from my bench and cannot try for a bit).

What I was getting from the previous comments on this thread was that it is a problem either in hardware or in the PicoSDK - as [https://github.com/raspberrypi/pico-sdk/issues/1144](issue #1144) seemed to indicate.

Note: to answer a comment in the code // what is this : 
`fmt.Printf("\033[2J") //  What is this?  <== escape sequence to clear the screen (xterm, terminal emulators)
fmt.Printf("\033[H")  //  What is this?  <== escape sequence to move the cursor to the top-left of the screen`

jceaser, 2026-10-15:
I have also seen similar issues and gave up trying to send anything that required accurate text like unix terminal codes.

rdon-key, 2026-10-16:
@jceaser 
The USB CDC implementation has changed significantly since this issue was reported.
Does this issue still occur with TinyGo 0.43?
I tested the original reproducer on both RP2040 and RP2350, but could not reproduce the issue.
If it still occurs, could you provide an updated reproducer and the test environment?