tinygo-org/tinygo #4888: TinyGo Not Working with pico2w RP2350 Chip

ZeyTroX-exe, 2025-05-06:
I'm trying to use TinyGo on a pico2w with the RP2350 chip, and I'm encountering issues. I've followed all the correct installation steps, including setting up TinyGo, but the program does not run correctly on the hardware. Below are the steps I've followed and the issues I'm facing.

Installed TinyGo following the official installation guide for my platform.

Installed dependencies and made sure the target is set correctly.

Tried flashing the program with the command:

tinygo flash -target=pico2-w main.go

go`
package main

import (
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.High()
		time.Sleep(time.Second / 2)

		led.Low()
		time.Sleep(time.Second / 2)
	}
}
`

Program does not execute as expected, the onboard LED on the Pico does not blink, and there is no serial output for debug information.

GeoffreyPlitt, 2025-05-06:
Not sure about the LED< but to see output from `println()` statements, run with `tinygo flash -monitor -target=pico2-w main.go`

SnapshotCiTy, 2025-05-07:
The led on the pico2 W is not wired to the same pin as on the pico2, it is wired to the wireless chip.

As far as I know you need to use the examples from this repository
https://github.com/soypat/cyw43439

examples/blinky worked on my pico2-w

this package contains drivers / programs and examples that deal specifically with the cyw43439 bluetooth/wifi chip of the pico-w and pico2-w

hb9cwp, 2025-05-11:
@ZeyTroX-exe I can confirm that [blinky.go](https://github.com/soypat/cyw43439/blob/main/examples/blinky/blinky.go) makes the LED blink via the CYW chip's own GPIO pins, which are separate from the GPIOs of the RP2350 chip, as pointed out by @SnapshotCiTy .

Also, look for the "WL_GPIO0" (in red, prefix "WL_" for WireLess) in the [schematic drawing of the Pico W](https://files.waveshare.com/wiki/Raspberry%20Pi%20Pico%202%20W/Pico-2-w-schematic.pdf).

dgryski, 2026-01-08:
Is this just a documentation issue then?  Is there anything we should add from the `cyw43439` repo to make our examples work out-of-the-box on a a pico2w?

micartey, 2026-01-19:
So I've run just now into this problem and as a dumb user I have two issues with this:

1. In my humble opinion, the `machine.LED` method should work - or throw an error
2. The documentation needs an update as it does not work. Even the blinky example uses a third party dependency...

jceaser, 2026-10-15:
this code does not work for me when I switch to GPIO28 and running through the clockwork picocalc hardware. My code works as normal when using a 2040 pico, but I recently switched over to the pico2-w and now existing code is not working as expected. To me, it seams as if time.Sleep() never returns. The led turns on, but never off. I have sound code that also uses sleep, the note will start, but never stop. Timing seams completely broken (to my eyes). `tinygo build -o tinygo-led2.uf2 -target=pico2-w led.go`. Again, exact code as the original post, I only swapped GPIO. I have a led with a resistor connected across ground and 28.

`> tinygo version`
`tinygo version 0.43.0 darwin/arm64 (using go version go1.25.14 and LLVM version 22.1.4)`

None of these packages show up in `brew outdated`.

davecheney, 2026-10-26:
The onboard LED is working on a Pico 2 W with TinyGo 0.44-dev using the CYW43439 driver. I flashed the following sample and confirmed the LED blinks once per second with the default scheduler.

```go
package main

import (
    "time"

    "github.com/soypat/cyw43439"
)

func main() {
    dev := cyw43439.NewPicoWDevice()
    if err := dev.Init(cyw43439.DefaultWifiConfig()); err != nil {
        panic(err)
    }
    for {
        if err := dev.GPIOSet(0, true); err != nil {
            panic(err)
        }
        time.Sleep(time.Second)
        if err := dev.GPIOSet(0, false); err != nil {
            panic(err)
        }
        time.Sleep(time.Second)
    }
}
```

In a new directory, save this as `main.go`, then run:

```sh
go mod init pico2w-blink
go get github.com/soypat/cyw43439@v0.2.0
tinygo flash -target=pico2-w .
```

The onboard LED is connected to the wireless chip, so it needs that driver rather than `machine.LED`. The separate report of `time.Sleep` hanging is tracked in #5898.

davecheney, 2026-11-08:
Thank you for your question. Please feel free to reopen this issue if you have additional questions.

closed 2026-11-08 by davecheney