tinygo-org/tinygo #4891: Troubles getting Pro Micro to run

JuLi0n21, 2025-05-10:
Hello TinyGo team,

Im currently struggling getting this [Pro Micro](https://de.aliexpress.com/item/1005006686008568.html) to run. ([datasheet](https://5.imimg.com/data5/SELLER/Doc/2020/10/RC/MV/IA/1833510/arduino-pro-micro-5v-16m-compatible.pdf))

I managed to flash to the mcu using the `arduino-leonardo` configuration but it seems to be doing nothing at all. 

i started with a blinky script migrated from code written in arduino IDE, which also detects the board as arduino-leonardo.

i saw and experienced the same issue described in this [issue](https://github.com/tinygo-org/tinygo/issues/4071)

tinygo code:
```
package main

import (
	"machine"
	"time"
)

func main() {
	rxLed := machine.Pin(17)
	txLed := machine.Pin(30)

	rxLed.Configure(machine.PinConfig{Mode: machine.PinOutput})
	txLed.Configure(machine.PinConfig{Mode: machine.PinOutput})

	for {
		rxLed.High()
		txLed.Low()
		time.Sleep(300 * time.Millisecond)

		rxLed.Low()
		txLed.High()
		time.Sleep(300 * time.Millisecond)
	}
}
```

original arduino code:
```
int RXLED = 17; 
int TXLED = 30; 

void setup() {
  
  pinMode(RXLED, OUTPUT);
  pinMode(TXLED, OUTPUT);
}

void loop() {
  digitalWrite(RXLED, HIGH);
  digitalWrite(TXLED, LOW);
  delay(300);
  digitalWrite(RXLED, LOW);
  digitalWrite(TXLED, HIGH);
  delay(300);
}
```

flash command: 
`tinygo flash -target arduino-leonardo -no-debug main.go`

i would be willing to create the missing pieces but i honestly dont even know where to start

sago35, 2025-05-11:
@jespino 
How was this part when you were developing it? It seems that writing to the `atmega32u4` on an `arduino-leonardo` removes the `USBCDC`, making recovery difficult.

JuLi0n21, 2025-05-12:
I also found this where some more [differences](https://forum.arduino.cc/t/pro-micro-5v-16mhz-atmega-32u4/113468) are noted

davecheney, 2026-10-26:
Thanks for reporting this. I have tested a Freetronics LeoStick, which also uses an ATmega32U4 at 16 MHz with Leonardo pin mapping, using TinyGo 0.44-dev.

Flashing and readback verification succeeded, but the user LED on PC7 remained solid rather than blinking. The same happened with a busy-wait delay instead of time.Sleep, and after unplugging and reconnecting the board. This looks like a bug, though the cause is not yet established.

The original example also uses Arduino numeric pin IDs, which are not equivalent to TinyGo machine.Pin values. Our test used the named MCU pin PC7 to avoid that difference. USB CDC support remains a separate issue tracked in #4071.

I will investigate further for 0.45.