tinygo-org/tinygo #4893: Very slow recompilation even for zero changes

gggrafff, 2025-05-14:
Hello! 

My service works with tinygo in this workflow:

- There is a pretty big project
- The request contains a piece of code for this project.
- We replace only one file in the project, compile it to wasi, and return the result.
- We receive a new request, change one file in the project, compile and give the result.
- And so on

Even if only one file has been replaced, recompilation takes a very long time, almost as long as the first compilation after tinygo clean.

I've been thinking for a long time now, is there any way I can speed this up? For example, if I split a project into modules, compile and link modules that don't change in advance, and then somehow compile and link one modifiable module to them? Or force tinygo to reuse the results of previous compilations and links more efficiently? Any ideas?

Thank you very much!

gggrafff, 2025-05-14:
The time for one compilation is about 3 minutes:

```
$ tinygo clean
$ time make 2>&1 | ts -m
May 14 13:40:25 tinygo build -x -tags wasm -o main.wasm -target ./target.json -opt=2 -gc=precise -scheduler=none -buildmode=wasi-legacy .
May 14 13:41:01 clang -Werror -Wall -std=c11 -nostdlibinc -c -Oz -gdwarf-4 -ffunction-sections -fdata-sections -Wno-macro-redefined --target=wasm32-unknown-wasi -fdebug-prefix-map=/tmp/tinygo1453700032/build-lib-compiler-rt=/tmp/tinygo-compiler-rt -resource-dir=/usr/local/lib/tinygo/lib/clang -mcpu=generic -o /tmp/tinygo1453700032/build-lib-compiler-rt/absvdi2.c.o /usr/local/lib/tinygo/lib/compiler-rt-builtins/absvdi2.c
May 14 13:41:02 clang -Werror -Wall -std=c11 -nostdlibinc -c -Oz -gdwarf-4 -ffunction-sections -fdata-sections -Wno-macro-redefined --target=wasm32-unknown-wasi -fdebug-prefix-map=/tmp/tinygo1453700032/build-lib-compiler-rt=/tmp/tinygo-compiler-rt -resource-dir=/usr/local/lib/tinygo/lib/clang -mcpu=generic -o /tmp/tinygo1453700032/build-lib-compiler-rt/absvsi2.c.o /usr/local/lib/tinygo/lib/compiler-rt-builtins/absvsi2.c
May 14 13:41:02 clang -Werror -Wall -std=c11 -nostdlibinc -c -Oz -gdwarf-4 -ffunction-sections -fdata-sections -Wno-macro-redefined --target=wasm32-unknown-wasi -fdebug-prefix-map=/tmp/tinygo1453700032/build-lib-compiler-rt=/tmp/tinygo-compiler-rt -resource-dir=/usr/local/lib/tinygo/lib/clang -mcpu=generic -o /tmp/tinygo1453700032/build-lib-compiler-rt/absvti2.c.o /usr/local/lib/tinygo/lib/compiler-rt-builtins/absvti2.c
May 14 13:41:02 clang -Werror -Wall -std=c11 -nostdlibinc -c -Oz -gdwarf-4 -ffunction-sections -fdata-sections -Wno-macro-redefined --target=wasm32-unknown-wasi -fdebug-prefix-map=/tmp/tinygo1453700032/build-lib-compiler-rt=/tmp/tinygo-compiler-rt -resource-dir=/usr/local/lib/tinygo/lib/clang -mcpu=generic -o /tmp/tinygo1453700032/build-lib-compiler-rt/adddf3.c.o /usr/local/lib/tinygo/lib/compiler-rt-builtins/adddf3.c
May 14 13:41:02 clang -Werror -Wall -std=c11 -nostdlibinc -c -Oz -gdwarf-4 -ffunction-sections -fdata-sections -Wno-macro-redefined --target=wasm32-unknown-wasi -fdebug-prefix-map=/tmp/tinygo1453700032/build-lib-compiler-rt=/tmp/tinygo-compiler-rt -resource-dir=/usr/local/lib/tinygo/lib/clang -mcpu=generic -o /tmp/tinygo1453700032/build-lib-compiler-rt/addsf3.c.o /usr/local/lib/tinygo/lib/compiler-rt-builtins/addsf3.c
May 14 13:41:02 clang -Werror -Wall -std=c11 -nostdlibinc -c -Oz -gdwarf-4 -ffunction-sections -fdata-sections -Wno-macro-redefined --target=wasm32-unknown-wasi -fdebug-prefix-map=/tmp/tinygo1453700032/build-lib-compiler-rt=/tmp/tinygo-compiler-rt -resource-dir=/usr/local/lib/tinygo/lib/clang -mcpu=generic -o /tmp/tinygo1453700032/build-lib-compiler-rt/addvdi3.c.o /usr/local/lib/tinygo/lib/compiler-rt-builtins/addvdi3.c
May 14 13:41:02 clang -Werror -Wall -std=c11 -nostdlibinc -c -Oz -gdwarf-4 -ffunction-sections -fdata-sections -Wno-macro-redefined --target=wasm32-unknown-wasi -fdebug-prefix-map=/tmp/tinygo1453700032/build-lib-compiler-rt=/tmp/tinygo-compiler-rt -resource-dir=/usr/local/lib/tinygo/lib/clang -mcpu=generic -o /tmp/tinygo1453700032/build-lib-compiler-rt/addvsi3.c.o /usr/local/lib/tinygo/lib/compiler-rt-builtins/addvsi3.c
May 14 13:41:02 clang -Werror -Wall -std=c11 -nostdlibinc -c -Oz -gdwarf-4 -ffunction-sections -fdata-sections -Wno-macro-redefined --target=wasm32-unknown-wasi -fdebug-prefix-map=/tmp/tinygo1453700032/build-lib-compiler-rt=/tmp/tinygo-compiler-rt -resource-dir=/usr/local/lib/tinygo/lib/clang -mcpu=generic -o /tmp/tinygo1453700032/build-lib-compiler-rt/addvti3.c.o /usr/local/lib/tinygo/lib/compiler-rt-builtins/addvti3.c
May 14 13:41:02 clang -Werror -Wall -std=c11 -nostdlibinc -c -Oz -gdwarf-4 -ffunction-sections -fdata-sections -Wno-macro-redefined --target=wasm32-unknown-wasi -fdebug-prefix-map=/tmp/tinygo1453700032/build-lib-compiler-rt=/tmp/tinygo-compiler-rt -resource-dir=/usr/local/lib/tinygo/lib/clang -mcpu=generic -o /tmp/tinygo1453700032/build-lib-compiler-rt/apple_versioning.c.o /usr/local/lib/tinygo/lib/compiler-rt-builtins/apple_versioning.c
May 14 13:41:02 clang -Werror -Wall -std=c11 -nostdlibinc -c -Oz -gdwarf-4 -ffunction-sections -fdata-sections -Wno-macro-redefined --target=wasm32-unknown-wasi -fdebug-prefix-map=/tmp/tinygo1453700032/build-lib-compiler-rt=/tmp/tinygo-compiler-rt -resource-dir=/usr/local/lib/tinygo/lib/clang -mcpu=generic -o /tmp/tinygo1453700032/build-lib-compiler-rt/ashldi3.c.o /usr/local/lib/tinygo/lib/compiler-rt-builtins/ashldi3.c
May 14 13:41:02 clang -Werror -Wall -std=c11 -nostdlibinc -c -Oz -gdwarf-4 -ffunction-sections -fdata-sections -Wno-macro-redefined --target=wasm32-unknown-wasi -fdebug-prefix-map=/tmp/tinygo1453700032/build-lib-compiler-rt=/tmp/tinygo-compiler-rt -resource-dir=/usr/local/lib/tinygo/lib/clang -mcpu=generic -o /tmp/tinygo1453700032/build-lib-compiler-rt/ashlti3.c.o /usr/local/lib/tinygo/lib/compiler-rt-builtins/ashlti3.c
May 14 13:41:02 clang -Werror -Wall -std=c11 -nostdlibinc -c -Oz -gdwarf-4 -ffunction-sections -fdata-sections -Wno-macro-redefined --target=wasm32-unknown-wasi -fdebug-prefix-map=/tmp/tinygo1453700032/build-lib-compiler-rt=/tmp/tinygo-compiler-rt -resource-dir=/usr/local/lib/tinygo/lib/clang -mcpu=generic -o /tmp/tinygo1453700032/build-lib-compiler-rt/ashrdi3.c.o /usr/local/lib/tinygo/lib/compiler-rt-builtins/ashrdi3.c
[... cut ...] b4ce2fbbb97c0ba5a2839f4c25d8f044e87893b94f3d4604d5bad924.bc /usr/local/lib/tinygo/lib/wasi-libc/sysroot/lib/wasm32-wasi/libc.a -mllvm -mcpu=generic -mllvm -mattr=+bulk-memory,+mutable-globals,+nontrapping-fptoint,+sign-ext,-multivalue,-reference-types --lto-O2 --thinlto-cache-dir=/home/ants/.cache/tinygo/thinlto
May 14 16:06:00 /usr/local/lib/tinygo/bin/wasm-opt -O2 -g /tmp/tinygo4217646321/main --output /tmp/tinygo4217646321/main.wasmopt

real    3m6,301s
user    2m41,583s
sys     0m18,016s
```

aykevl, 2025-05-24:
This is something which I want to fix long-term, but it is much more complicated than it might seem. We have a few whole-program compilation passes that are currently required, they can't be switched off. The goal would be to replace these passes with package-level passes and compile every package as one object file (and link those), but doing that will likely increase binary size in many cases.

I do wonder why you need to recompile the whole binary with every request?

davecheney, 2026-10-26:
Sorry for the delay. Could you retry with TinyGo 0.43 or newer and capture a CPU profile using `-cpuprofile=compile.prof`? Please include your TinyGo version, build command, custom target file, and timings for two consecutive builds without running `tinygo clean` between them. A small project demonstrating the slowdown would also help.