tinygo-org/tinygo #4894: TypeScript Go WASM

sxzz, 2025-05-15:
I tested the latest development commit (2da9d26e21fefc68eaab98fea30833e024a38bf5) of TinyGo to compile [TypeScript Go](https://github.com/microsoft/typescript-go).

The previous issue (https://github.com/tinygo-org/tinygo/issues/4804) has been resolved; however, there are still some remaining issues:

- `os.Getwd()` always returns an empty string, which causes a panic in tsgo.
- The PR https://github.com/tinygo-org/tinygo/pull/4565 are very useful; we need this functionality for CLI arguments. /cc @aykevl

aykevl, 2025-05-21:
> `os.Getwd()` always returns an empty string, which causes a panic in tsgo.

I don't think wasip1 has a concept of current working directory, but we can of course return a fake directory like `"."`.

> The PR at https://github.com/tinygo-org/tinygo/pull/4565 is very useful; we need this functionality for CLI arguments.

It doesn't change anything for wasip1. For wasip1, command line arguments are already supported.

sxzz, 2025-05-21:
In the WASM (non-WASI) build of Golang, `os.Getwd()` internally calls `process.cwd()`. Currently, only the WASM version is compatible with tsgo; using the WASI version will result in a `RangeError: Maximum call stack size exceeded`.

aykevl, 2025-05-21:
> `os.Getwd()` always returns an empty string, which causes a panic in tsgo.

I checked, and it returns `"/"` and not an empty string.

aykevl, 2025-05-21:
https://github.com/tinygo-org/tinygo/pull/4901 refactors signal handling, and adds `os/signal.signalWaitUntilIdle`. `tsgo` now compiles to wasip1 in TinyGo!

...however, it doesn't really work yet. It does show help output, but it crashes when actually compiling anything. There seems to be a memory corruption somewhere (`checker` inside `CheckSourceFiles` of program.go becomes nil when it shouldn't).

dgryski, 2026-01-08:
Current issues building typescript-go:
 -  https://github.com/tinygo-org/tinygo/issues/5040 (missing os.UserConfigDir)

```
~/go/src/github.com/microsoft/typescript-go $ tinygo build ./cmd/tsgo
# github.com/microsoft/typescript-go/internal/format
internal/format/rulecontext.go:19:2: generic type alias requires GODEBUG=gotypesalias=1 or unset
internal/format/rulecontext.go:19:70: undefined: T
~/go/src/github.com/microsoft/typescript-go $ (GODEBUG=gotypesalias=1 tinygo build ./cmd/tsgo)
# github.com/microsoft/typescript-go/cmd/tsgo
cmd/tsgo/lsp.go:84:22: undefined: os.UserCacheDir
cmd/tsgo/lsp.go:86:25: undefined: os.UserConfigDir
~/go/src/github.com/microsoft/typescript-go $ (GODEBUG=gotypesalias=1 tinygo build ./cmd/tsgo)
panic: got github.com/go-json-experiment/json/jsontext.Value, want variadic parameter of unnamed slice or string type
```