tinygo-org/tinygo #4918: xiao-ble (sense), nrf52840 misleading I2C0 comment

gen2thomas, 2025-06-06:
The comment [`// I2C0 (external) pins`](https://github.com/tinygo-org/tinygo/blob/3e60eeb368f25f237a512e7553fd6d70f36dc74c/src/machine/board_xiao-ble.go#L72C2-L72C25) is misleading, because although configuring the bus with `machine.I2C0.Configure(machine.I2CConfig{})` the external pins will not be used. The reason seems to be the [same code](https://github.com/tinygo-org/tinygo/blob/3e60eeb368f25f237a512e7553fd6d70f36dc74c/src/machine/machine_nrf.go#L252C1-L256C3) for all "I2Cx" together with the [defaulting to internal](https://github.com/tinygo-org/tinygo/blob/3e60eeb368f25f237a512e7553fd6d70f36dc74c/src/machine/board_xiao-ble.go#L68C1-L70C20).

To get the external pins to work you need to set the pins explicitly like this:  `machine.I2C0.Configure(machine.I2CConfig{SCL: machine.SCL0_PIN, SDA: machine.SDA0_PIN})`.

deadprogram, 2025-06-08:
Should we change the comment, or change the default values?

gen2thomas, 2025-06-08:
It would be nice, if the comment will be implemented. This means I2C0 would have the external pins per default and the I2C1 would have the internal pins per default. The internal pins are needed to access the IMU (LSM6DS3TR) in the "Sense"-version of the MCU. For the normal version only the external pins are needed.

This suggestion leads to a small problem with the existing [example for IMU-driver](https://github.com/tinygo-org/drivers/blob/release/examples/lsm6ds3tr/main.go), which uses the I2C0.

deadprogram, 2025-06-09:
> This suggestion leads to a small problem with the existing [example for IMU-driver](https://github.com/tinygo-org/drivers/blob/release/examples/lsm6ds3tr/main.go), which uses the I2C0

So what is the best solution for the most users, do you think?

gen2thomas, 2025-06-09:
Because the example is used for all MCU's and the xiao-ble sense is just one exception, I think implementing my suggestion would help the users of xiao-ble with and without sense a lot. A comment in the example could close the gap.

I have no knowledge about the "Seeed XIAO ESP32 Sense" variant at all.