tinygo-org/tinygo #4931: Something not OK

d-enk, 2025-06-20:
tinygo version 0.38.0 linux/amd64 (using go version go1.24.4 and LLVM version 19.1.2)

Managed to simplify it to such a reproducer

```go
-- go.mod --
module test

go 1.18
-- test_test.go --

package main

import (
	"testing"
)

type Label struct{}

type Item[T any] struct {
	Value  T
	Labels []Label
}

func run[T any](input <-chan Item[T]) {
	for item := range input {
		if len(item.Labels) != 0 {
			panic(len(item.Labels))
		}
	}
}

func NotUsed() {
	type T struct{ any }

	run[T](nil)
}

func Benchmark(b *testing.B) {
	type T struct{}

	items := make(chan Item[T])

	go func() {
		for i := 0; i < b.N; i++ {
			items <- Item[T]{Labels: nil}
		}

		close(items)
	}()

	run(items)
}
```

`tinygo test -benchmem -run=^$ -bench ^ -v -count=1`
```
panic: 2284738
FAIL  test  1.149s
```

Gustavomurta, 2025-06-29:
What microcontroller are you using?
Why go 1.18? 

-- go.mod --
module test
**go 1.18**

d-enk, 2025-06-30:
> What microcontroller are you using?

Nothing
Ubuntu 24.04.2 LTS [x86_64] Linux 6.11.0-26-generic
Intel(R) Core(TM) i3-8100 CPU @ 3.60GHz x 4

> Why go 1.18?

Just the minimum required version.

aykevl, 2025-08-20:
I can reproduce this issue. There's certainly something weird going on.

aykevl, 2025-08-20:
The name of `T` is relevant, renaming it hides the bug.
My guess is something relies on the uniqueness of type names.

aykevl, 2025-08-20:
I think the problem is that there are two functions called `run[command-line-arguments.T]` where `T` is different for each (but the name is the same).

aykevl, 2025-08-20:
The documentation for `RelString` says:

> All non-synthetic functions have distinct package-qualified names. (But two methods may have the same name “(T).f” if one is a synthetic wrapper promoting a non-exported method “f” from another package; in that case, the strings are equal but the identifiers “f” are distinct.)

...and we were relying on the uniqueness of these names :/