tinygo-org/tinygo #4949: Documentation: GC on WASM

ShalokShalom, 2025-07-09:
Hi there 🙂 

Is the memory usage of the new GC still so bad as for the old one?
Your docs have not been updated in a while, and I found several improvements in the release notes. Both related to the new GC, and to memory usage at all. 

Can we revisit the current wording in the documentation about GC on wasm?
https://tinygo.org/docs/reference/lang-support/

P.S: I tried reaching you in chat, but Slack is horrible as always, and offers me no registration option. Just login. Do you have sane contact methods?

ShalokShalom, 2025-12-16:
So its better, but still bad?

> ... may work not as well on very small chips (AVR) and on WebAssembly. It is also a lot slower than the usual Go garbage collector.

Thanks a lot

dgryski, 2025-12-16:
If I were to make any change to the docs, I would replace "a lot slower" with simply "slower".   With the inclusion of bdwgc, `-gc=leaking`, and the improvements to tinygo's own conservative collector I don't think it's fair to call it "a lot slower" any more.

ShalokShalom, 2025-12-17:
Sounds sensible