tinygo-org/tinygo #4959: wasm-ld: error: couldn't allocate output register for constraint '{r0}'

anshuman-mor, 2025-07-17:
Getting this error intermittently when compiling to wasip2

> tinygo build -o test-tinygo-wasi2.wasm --target wasip2 --no-debug

wasm-ld: error: couldn't allocate output register for constraint '{r0}'

I am running this on Windows.

> tinygo version 
tinygo version 0.38.0 windows/amd64 (using go version go1.23.2 and LLVM version 19.1.2)

deadprogram, 2025-08-12:
Hello @anshuman-mor could you please provide some code that reproduces this error? Thank you.

joelnpdr, 2025-08-20:
Hello, I have the same issue.  I am compiling this library, which has some crypto operations:

https://github.com/gagliardetto/solana-go

aykevl, 2025-09-24:
Oh, huh. Somehow this program is calling syscalls, which of course won't work on wasip2. I wonder how that happened.

aykevl, 2026-04-16:
The original issue has been fixed - sort of. So closing. If there are problems with the current approach of not allowing arm syscalls on wasm, please create a new issue.

> Side note: is there a list somewhere of what standard library packages we gain by "emulating" linux/arm, over just setting GOOS/GOARCH correctly?

There's no such list, but we absolutely need to pick one of the existing GOOS/GOARCH combinations. I don't remember the specific issue why we needed linux/arm for wasm, probably just the lack of https://github.com/golang/go/issues/63131 (which we effectively use).

closed 2026-04-16 by aykevl