tinygo-org/tinygo #4965: goroutines.go may be flaky

QuLogic, 2025-07-20:
Once it failed as `TestBuild/ARMLinux/goroutines.go`, and once as `TestBuild/X86Linux/goroutines.go`, but it was the same diff:
```
=== NAME  TestBuild/ARMLinux/goroutines.go
    main_test.go:874: output did not match (expected 523 bytes, got 523 bytes):
    main_test.go:875: diff expected actual
        --- expected
        +++ actual
        @@ -2,8 +2,8 @@
         goroutine in init
         main 1
         sub 1
        -main 2
         sub 2
        +main 2
         main 3
         wait:
           wait start
        
    main_test.go:477: stdout: init
    main_test.go:477: stdout: goroutine in init
    main_test.go:477: stdout: main 1
    main_test.go:477: stdout: sub 1
    main_test.go:477: stdout: sub 2
    main_test.go:477: stdout: main 2
    main_test.go:477: stdout: main 3
    main_test.go:477: stdout: wait:
    main_test.go:477: stdout:   wait start
    main_test.go:477: stdout:   wait end
    main_test.go:477: stdout: end waiting
    main_test.go:477: stdout: value produced after some time: 42
    main_test.go:477: stdout: non-blocking goroutine
    main_test.go:477: stdout: done with non-blocking goroutine
    main_test.go:477: stdout: async interface method call
    main_test.go:477: stdout: slept inside func pointer 8
    main_test.go:477: stdout: slept inside closure, with value: 20 8
    main_test.go:477: stdout: closure go call result: 1
    main_test.go:477: stdout: pre-acquired mutex
    main_test.go:477: stdout: releasing mutex
    main_test.go:477: stdout: acquired mutex from goroutine
    main_test.go:477: stdout: releasing mutex from goroutine
    main_test.go:477: stdout: re-acquired mutex
    main_test.go:477: stdout: done
    main_test.go:477: stdout: called: Foo.Nowait
    main_test.go:477: stdout: called: Foo.Wait
    main_test.go:477: stdout:   ...waited
    main_test.go:477: stdout: done with 'go on interface'
    main_test.go:477: stdout: paddedStruct: 5 7
```
I did apply #4958, but that is in a later part of the test, so didn't fix it.

eliasnaur, 2025-07-21:
FWIW, I de-flaked this test in https://github.com/tinygo-org/tinygo/pull/4958. It seems to me prints may introduce scheduling points and as such need similar channel-based synchronization.

QuLogic, 2025-08-02:
Yes, I applied that patch, but it's unfortunately not enough.

eliasnaur, 2025-08-03:
What I meant was that a similar fix will be needed for this issue, not that #4958 fixes it.