tinygo-org/tinygo #4969: TestBuild/X86Linux fails on aarch64

QuLogic, 2025-07-23:
Of course, this is running with qemu since it's cross-architecture, but I believe this used to work in the past. Now it fails with something to do with bdwgc:
```
=== CONT  TestBuild/X86Linux/filesystem.go
GC Warning: Out of memory - trying to allocate requested amount (4112 bytes)...
Insufficient memory for GC_all_nils
```

aykevl, 2025-09-29:
Sounds like a QEMU issue (though it's still on us to fix, probably).
I think I've seen the same issue when running locally (I'm running Fedora Asahi). We don't test for this in CI, since (almost) all our CI is running amd64 and not 386.

davecheney, 2026-10-28:
Reproduced on an AMD64 Linux host with QEMU 11.1.2 and TinyGo 0.44.0-dev. ARM64 is not required.

```sh
GOOS=linux GOARCH=386 tinygo build -gc=boehm -o filesystem ./testdata/filesystem.go
qemu-i386 ./filesystem
```

This fails with the same error:

```text
GC Warning: Out of memory - trying to allocate requested amount (4112 bytes)...
Insufficient memory for GC_all_nils
```

The identical binary runs successfully without QEMU on the same host.

davecheney, 2026-11-08:
This problem is not a TinyGo regression. The cause is how QEMU user mode and the bdwgc mmap address hint work together.

On Linux/i386, bdwgc sets `HEAP_START` to `0x1000` ([gcconfig.h](https://github.com/bdwgc/bdwgc/blob/master/include/private/gcconfig.h), `I386` and `LINUX` section). bdwgc gives this value to `mmap` as an address hint.

The Linux kernel does not use a hint that is below `vm.mmap_min_addr`. The kernel selects a different address. Thus, the binary runs correctly without QEMU.

On a 64-bit host, `qemu-i386` always uses `reserved_va`. In this mode, QEMU finds a free guest range that starts at the hint. Then QEMU changes the mapping to `MAP_FIXED` (`linux-user/mmap.c`, functions `mmap_find_vma_reserved` and `target_mmap__locked`). QEMU does not increase the hint to `mmap_min_addr`.

When `guest_base` is 0, the host mapping is at address `0x1000`. The host kernel does not permit this mapping and returns `EPERM`:

    $ qemu-i386 -strace ./filesystem
    mmap2(0x00001000,65536,PROT_READ|PROT_WRITE,MAP_PRIVATE|MAP_ANONYMOUS,-1,0) = -1 errno=1 (Operation not permitted)
    GC Warning: Out of memory - trying to allocate requested amount (4112 bytes)...
    mmap2(0x00001000,8192,PROT_READ|PROT_WRITE,MAP_PRIVATE|MAP_ANONYMOUS,-1,0) = -1 errno=1 (Operation not permitted)
    Insufficient memory for GC_all_nils

    $ qemu-i386 -d page ./filesystem 2>&1 | grep guest_base
    guest_base  (nil)

If you set a guest base that is not 0, the test passes. The output is the same as `testdata/filesystem.txt`:

    $ QEMU_GUEST_BASE=0x10000000 qemu-i386 ./filesystem
    $ qemu-i386 -B 0x10000000 ./filesystem

CI does not find this problem. On amd64 hosts, `compileopts/target.go` runs linux/386 binaries without QEMU. Only hosts with a different architecture use `qemu-i386`.

Possible fixes:

1. QEMU: In the `reserved_va` path, increase a hint that is not fixed to a minimum of `mmap_min_addr`. The kernel does this. This is the correct fix.
2. bdwgc: For Linux/i386, use a hint of `0x10000` or higher. We cannot change the value with `-D` in `builder/bdwgc.go`, because the header does not have an `#ifndef` guard.
3. TinyGo: Add a guest base to the `qemu-i386` emulator command. This fix is simple, but it puts a fixed address in the code.

Test environment: QEMU 11.1.3, TinyGo 0.45.0-dev, LLVM 22.1.8, amd64 Arch Linux host.

closed 2026-11-08 by deadprogram