tinygo-org/tinygo #4970: Raspberry Pico running at 200MHz - non-standard

Gustavomurta, 2025-07-24:
When testing with my Raspberry Pico boards, I found that they were all set to 200 MHz when I compile Tinygo programs.

I identified this change in Tinygo:

Gustavomurta, 2025-07-27:
Hi eliasnaur, 
When you purchase a Pico board, it comes with a 125 MHz clock, right? In my opinion, the 200 MHz clock could be optional.

In my case, my test applications depend on the clock frequency. Therefore, I've already developed the program to change the clock. So this 200 MHz option is up to the Tinygo developers. Thank you.

eliasnaur, 2025-07-27:
I don't know what you mean by "comes with a 125MHz clock". The clock speed is configurable by software, and if anything, the maximum clock speed was 133MHz before being bumped to 200MHz.

Perhaps by "comes with a 125MHz clock" you mean that TinyGo used to run Pico boards at 125MHz and now runs them at a different speed. That's true, but I don't think it's reasonable to expect `machine.CPUFrequency` to remain constant between rebuilds (for the same reason you can't expect `runtime.NumCPU` to stay constant - and indeed the latest TinyGo did bump rp2040's `runtime.NumCPU` to 2).

That leaves the question whether changing clock speeds should be offered as an option. Do you have a compelling use case for such an option?

Gustavomurta, 2025-07-27:
Dear Eliasnur,
Just for reference, the **Arduino-pico IDE (earlephilhower)** offers numerous clock options. Just a suggestion.

https://github.com/earlephilhower/arduino-pico

[image]

My projects involve data capture at various speeds, including extreme overclocking.

xudongzheng, 2025-08-05:
> That said, is there a reason to offer a 125MHz option?

I have a custom RP2040 board though I'm building with `-target pico`.

With CPU frequency set to 200MHz, "hello world" doesn't even run. It seems to be fine at 150MHz, 160MHz is where it breaks. There is a hard fault according to GDB:

```
[rp2040.core0] clearing lockup after double fault
[rp2040.core0] halted due to debug-request, current mode: Handler HardFault
xPSR: 0x60000003 pc: 0xfffffffe msp: 0x200007e0
```

I haven't had a chance to investigate further yet.