tinygo-org/tinygo #4988: sync.Mutex pointer receiver issue

mikesmitty, 2025-08-08:
As I've been testing out the multi-core functionality with a project I'm working on I've run into an issue with sync.Mutex. If it's not explicitly configured as `*sync.Mutex` it will not be converted to a pointer by the pointer receiver on the Lock/Unlock methods and results in double-unlock panics:
> panic: sync: unlock of unlocked Mutex

I've worked around it in my code by just making all my mutexes explicitly pointer variables, but this one in particular is biting me while setting up two ADC channels in quick succession:

`var adcLock sync.Mutex`
https://github.com/tinygo-org/tinygo/blob/0ec1cb1e19b33d0d5f3ae788f26af0c28f46a24e/src/machine/machine_rp2_adc.go#L15

Apologies if that doesn't make sense, struggling with my words today. I came across this comment explaining the standard behavior that made realize what was going on, that the conversion of "a.m.Lock() to (&a.m).Lock()" is not happening:

> First of all you're using the A struct as a pointer receiver which is the recommended when holding sync.Mutex in the struct fields. If you check the sync.Mutex code you'll see that it's also a pointer receiver to the Lock() and Unlock() methods. Go will automatically convert the a.m.Lock() to (&a.m).Lock() as this is user friendly. You can be explicit and pass *sync.Mutex in the struct fields but that would mean that you'd also need to initialize it explicitly something := A{m: new(sync.Mutex)}. So it's simpler to just use the value to begin with and let Go do it's magic

https://www.reddit.com/r/golang/comments/u9o5wj/mutex_struct_field_as_reference_or_value/

mikesmitty, 2025-08-09:
Yeah, as I'm trying to make a simple reproduction of it my initial assessment is definitely wrong. I've got a bit of a large and inconvenient reproduction on this branch here: https://github.com/mikesmitty/beacon-dcc/tree/mutex-flaw-reproducer

It happens fairly reliably for me immediately after startup with that example, sometimes requiring a second boot after hitting the reset button then every time afterwards. It seems to be very sensitive and timing-based so I'm having a difficult time trimming it down to something more manageable.

mikesmitty, 2025-08-09:
As I dig further the panic is apparently a misnomer:
```
(gdb) c
Continuing.
^C
Thread 1 "rp2350.dap.core0" received signal SIGINT, Interrupt.
runtime.panicOrGoexit (message=..., panicking=<optimized out>)
    at /Users/michaelsmith/git/tinygo/src/runtime/panic.go:81
81              abort()
(gdb) bt
#0  runtime.panicOrGoexit (message=..., panicking=<optimized out>)
    at /Users/michaelsmith/git/tinygo/src/runtime/panic.go:81
#1  0x10001fca in (*runtime/volatile.Register32).HasBits (r=0x0, value=268496000)
    at /Users/michaelsmith/git/tinygo/src/runtime/volatile/register.go:181
#2  <signal handler called>
```

mikesmitty, 2025-09-08:
I believe so? It's set to 8k on the target I'm using, and if I add a sleep between starting the goroutine and the function that creates a new map it doesn't happen

eliasnaur, 2025-09-08:
One drive-by comment while spelunking nearby code:

Shouldn't `markedTaskQueue` be emptied when retrying at https://github.com/tinygo-org/tinygo/blob/3869f76887feef6c444308e7e1531b7cac1bbd10/src/runtime/gc_blocks.go#L494-L496?

aykevl, 2025-09-08:
> I'm assuming there's no implicit pointer indirection in TinyGo?

TinyGo follows the Go standard here, a deviation would be unexpected. (Not impossible, but such a big issue should have been found ages ago).

aykevl, 2025-09-08:
According to the stack trace this is where the lock gets unlocked while it is not locked: 

https://github.com/tinygo-org/tinygo/blob/3869f76887feef6c444308e7e1531b7cac1bbd10/src/runtime/gc_blocks.go#L409

I just find it a bit hard to believe, that code looks correct to me.

mikesmitty, 2025-09-08:
> > I'm assuming there's no implicit pointer indirection in TinyGo?
> 
> TinyGo follows the Go standard here, a deviation would be unexpected. (Not impossible, but such a big issue should have been found ages ago).

Yeah, my initial thought was way off-base. I've since done a lot more digging and my current pet theory is that an interrupt is firing between the CBNZ and STLEX of the `futex.CompareAndSwap()` here and the exclusivity lock is somehow getting timed out
```
   0x1000681c <runtime.alloc+40>:	ldaex	r1, [r5]
   0x10006820 <runtime.alloc+44>:	cbnz	r1, 0x1000683a <runtime.alloc+70>
   0x10006822 <runtime.alloc+46>:	stlex	r1, r0, [r5]
```
I tried adding futex.CompareAndSwap() to futex-cores.go that does `lockFutex()`/`unlockFutex()` around it and it didn't do it, but since it's very timing-dependent I'll need to do a lot more testing to see if it actually resolves it
```
func (f *Futex) CompareAndSwap(old, new uint32) (swapped bool) {
	mask := lockFutex()
	swapped = f.Uint32.CompareAndSwap(old, new)
	unlockFutex(mask)
	return
}
```

aykevl, 2025-09-08:
Are you sure you're looking at the right assembly? As far as I know the Cortex-M0+ of the RP2040 doesn't support ldaex/stlex.

davecheney, 2026-10-27:
Sorry for the delay. Could you retry your reproducer with TinyGo 0.43, or 0.44 when it ships in a few days? There have been fixes to the multicore scheduler since this report, but we have not confirmed whether they resolve the double-unlock panic. If it persists, please include your TinyGo version, board, build command, and updated stack traces.