tinygo-org/tinygo #5018: TinyGo example code causes Pico to frezze

Hangover1234, 2025-08-27:
I have tried running the example code for the max6675 temp sensor on a pi pico, resulting in the pico freezing and needing a reboot. I have tried changing the cables and switching the pico out, with no change. 

Some simple troubleshooting showed, that the program hangs when it tries to read the sensor.
Getting desperate, i changed the SDI to SDO, with no luck.

In the following code, I have switched the pins from adafruit to pico 

`func main() {
    machine.GPIO17.Configure(machine.PinConfig{Mode: machine.PinOutput})
    machine.GPIO17.High()

    machine.SPI0.Configure(machine.SPIConfig{
        Frequency: 1_000_000,
        SCK:       machine.GPIO18,
        SDO:       machine.GPIO16,
    })

    thermocouple := max6675.NewDevice(machine.SPI0, machine.GPIO17)

    for {
        temp, err := thermocouple.Read()

        if err != nil {
            println(err)
            return
        }
        println("%0.02f C : %0.02f F\n", temp, (temp*9/5)+32)
        time.Sleep(time.Second)
    }
}`
Any ideas what could cause this in the example code?

eliasnaur, 2025-08-28:
Perhaps this is a scheduler problem. Try the `-scheduler tasks` flag as noted in a potentially [related issue](https://github.com/tinygo-org/tinygo/issues/4974#issuecomment-3149328430). See also https://github.com/tinygo-org/tinygo/issues/4975.

Hangover1234, 2025-08-28:
I have now tried 
`tinygo flash -target=pico -scheduler tasks`
But no luck. 
I have also looked at the related issue and have tried using `gdb`, but I am not that well versed in it.
I get errors
`"monitor" command not supported by this target.
You can't do that when your target is `exec'
"monitor" command not supported by this target.`
Probably obvious errors, not sure if `gdb` is the solution.

deadprogram, 2025-09-10:
I do not have one of the specific sensors to test with. However, just eyeballing differences in your code vs the example.

The example configures the pin used for the CS as an output, and sets it on:

```go
machine.D5.Configure(machine.PinConfig{Mode: machine.PinOutput})
machine.D5.High()
```

You probably need to do the same on pin `GPIO17`.

As far as the SPI interface, I would suggest trying the defaults.

```go
machine.SPI0.Configure(machine.SPIConfig{})
```

Lastly, make sure you have the sensor connected to the correct pins. See https://datasheets.raspberrypi.com/pico/Pico-R3-A4-Pinout.pdf

For example, `GPIO17` is the pin numbered `22` on the Pico.

Hope that helps!

deadprogram, 2026-01-13:
Closing since this question appears to have been answered. Thanks everyone!

closed 2026-01-13 by deadprogram